cjdrake/pyeda #125: Another question about espresso_tts

E-Neo, 2015-08-07:
Excuse me, I have another question about Espresso

A, B, C = map(exprvar, 'ABC')
f_tt = truthtable((C, B, A), '10110101')
f_ex = truthtable2expr(f_tt)
g_ex = espresso_tts(f_tt)[0]
f_ex.equivalent(g_ex)

I think it should return True?

cjdrake, 2015-08-07:
Seems like a bug.

E-Neo, 2015-08-07:
It seems like espresso convert the permutation of the variables:

f_tt = truthtable((C, B, A), '10110101')
g_tt = expr2truthtable(espresso_tts(f_tt)[0])
f_tt
  A B C                                                                                                                       
  0 0 0 : 1                                                                                                                   
  0 0 1 : 0                                                                                                                   
  0 1 0 : 1                                                                                                                   
  0 1 1 : 1                                                                                                                   
  1 0 0 : 0                                                                                                                   
  1 0 1 : 1                                                                                                                   
  1 1 0 : 0                                                                                                                   
  1 1 1 : 1
f_tt = truthtable((D, C, B, A), '1011010110010100')
g_tt = expr2truthtable(espresso_tts(f_tt)[0])
f_tt
  A B C D
  0 0 0 0 : 1
  0 0 0 1 : 0
  0 0 1 0 : 1
  0 0 1 1 : 1
  0 1 0 0 : 0
  0 1 0 1 : 1
  0 1 1 0 : 0
  0 1 1 1 : 1
  1 0 0 0 : 1
  1 0 0 1 : 0
  1 0 1 0 : 0
  1 0 1 1 : 1
  1 1 0 0 : 0
  1 1 0 1 : 1
  1 1 1 0 : 0
  1 1 1 1 : 0
g_tt
  D C B A
  0 0 0 0 : 1
  0 0 0 1 : 0
  0 0 1 0 : 1
  0 0 1 1 : 1
  0 1 0 0 : 0
  0 1 0 1 : 1
  0 1 1 0 : 0
  0 1 1 1 : 1
  1 0 0 0 : 1
  1 0 0 1 : 0
  1 0 1 0 : 0
  1 0 1 1 : 1
  1 1 0 0 : 0
  1 1 0 1 : 1
  1 1 1 0 : 0
  1 1 1 1 : 0

E-Neo, 2015-08-07:
It seems like espresso converts the permutation of the variables.

f_tt = truthtable((D, C, B, A), '1011010110010100')
g_tt = expr2truthtable(espresso_tts(f_tt)[0])
f_tt
A B C D
0 0 0 0 : 1
0 0 0 1 : 0
0 0 1 0 : 1
0 0 1 1 : 1
0 1 0 0 : 0
0 1 0 1 : 1
0 1 1 0 : 0
0 1 1 1 : 1
1 0 0 0 : 1
1 0 0 1 : 0
1 0 1 0 : 0
1 0 1 1 : 1
1 1 0 0 : 0
1 1 0 1 : 1
1 1 1 0 : 0
1 1 1 1 : 0
g_tt
D C B A
0 0 0 0 : 1
0 0 0 1 : 0
0 0 1 0 : 1
0 0 1 1 : 1
0 1 0 0 : 0
0 1 0 1 : 1
0 1 1 0 : 0
0 1 1 1 : 1
1 0 0 0 : 1
1 0 0 1 : 0
1 0 1 0 : 0
1 0 1 1 : 1
1 1 0 0 : 0
1 1 0 1 : 1
1 1 1 0 : 0
1 1 1 1 : 0

E-Neo, 2015-09-05:
Maybe there is something wrong in the file /pyeda/pyeda/boolalg/espressomodule.c
because your espresso in the directory thirdparty/espresso gaves the correct answer but after wraped to python by espressomodule.c it goes wrong.

cjdrake, 2023-08-27:
danjujan, this fix looks good to me.

closed 2023-08-27 by cjdrake