cjdrake/pyeda #126: Can't Install

ChipJust, 2015-08-08:
I might be doing something wrong, but the C code won't compile for me...I just want to install the thing. I am using Win 8.1. I installed the Visual Studio for Python 2.7 that you referenced. I have a couple of other VS installs on this machine, don't know if that is causing an issue here.

Here is the build error. There are lots of warnings before this, by the way.
C:\Users\cmueltsc\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\amd64\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -Ithirdparty\espresso\src -IC:\Python27\include -IC:\Python27\PC /Tcpyeda\boolalg\espressomodule.c /Fobuild\temp.win-amd64-2.7\Release\pyeda\boolalg\espressomodule.obj espressomodule.c
pyeda\boolalg\espressomodule.c(59) : warning C4244: '=' : conversion from 'Py_ssize_t' to 'int', possible loss of data
pyeda\boolalg\espressomodule.c(67) : warning C4244: '=' : conversion from 'Py_ssize_t' to 'int', possible loss of data
pyeda\boolalg\espressomodule.c(112) : warning C4244: '=' : conversion from 'Py_ssize_t' to 'int', possible loss of data
pyeda\boolalg\espressomodule.c(498) : error C2061: syntax error : identifier '_module'
pyeda\boolalg\espressomodule.c(498) : error C2059: syntax error : ';'
pyeda\boolalg\espressomodule.c(498) : error C2513: '/*global*/ ' : no variable declared before '='
pyeda\boolalg\espressomodule.c(499) : error C2065: 'PyModuleDef_HEAD_INIT' : undeclared identifier
pyeda\boolalg\espressomodule.c(513) : warning C4013: 'PyModule_Create' undefined; assuming extern returning int
pyeda\boolalg\espressomodule.c(513) : error C2065: '_module' : undeclared identifier
pyeda\boolalg\espressomodule.c(513) : warning C4047: '=' : 'PyObject *' differs in levels of indirection from 'int'
pyeda\boolalg\espressomodule.c(539) : warning C4098: 'PyInit_espresso' : 'void' function returning a value
pyeda\boolalg\espressomodule.c(542) : warning C4098: 'PyInit_espresso' : 'void' function returning a value
error: command 'C:\\Users\\cmueltsc\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\amd64\\cl.exe' failed with exit status 2

ChipJust, 2015-08-08:
Ok, I just saw that you don't support 2.7. My bad, I thought you did. I have a different problem with 3.4.3 Compilation, but I don't think this is specific to you...
error: Microsoft Visual C++ 10.0 is required (Unable to find vcvarsall.bat).
I think this is a 64-bit Windows difficulty...
Do you have your pre-built binaries posted anywhere?

sschnug, 2015-08-08:
What about having a look into the official documentation [here](http://pyeda.readthedocs.org/en/latest/install.html)?
This should be easy to follow and it's working for me. (Don't compile it yourself!)

ChipJust, 2015-08-08:
Yea, fair enough.

closed 2015-08-08 by ChipJust