cjdrake/pyeda #138: AttributeError: 'XorOp' object has no attribute 'to_cdnf'

vlade11115, 2016-11-24:
Hi.

[Canonical Normal Form](http://pyeda.readthedocs.io/en/v0.28.0/expr.html#canonical-normal-form) in docs is broken now.
I'm trying to run the example from the documentation at the link above.
```>>> a, b, c = map(exprvar, ('a', 'b', 'c'))
>>> f = Xor(a, Implies(b, c))
>>> f
Xor(a, Implies(b, c))
>>> f.to_cdnf()
Traceback (most recent call last):
  File "/usr/lib/python3.5/code.py", line 91, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
AttributeError: 'XorOp' object has no attribute 'to_cdnf'
```
Same thing with another form (ccnf).
How can I get ccnf and cdnf forms?

I am using python3, Ubuntu 16.04, the latest version of the library (0.28.0) installed by pip. Other examples of work, such as to_dnf() or to_cnf() works good.

ZikDead, 2016-11-24:
On windows 10 I have the same problem.

cjdrake, 2016-11-27:
Thanks for the bug report. I just moved, so I'm a little late looking at stuff right now.

oscarys, 2019-09-07:
Hello everyone. I was trying to use *.to_ccnf() and this error persists (it seems that this code hasn't been touched for a long while). It is a great library, though !!

myzinsky, 2023-05-25:
+1 here...

myzinsky, 2023-05-25:
A quick workaround is to use expansion(all variables...) to get the canonical form.

cjdrake, 2023-05-26:
I wild Matthias Jung appeared :).

This comment sent me on a brief trip down memory lane. I got rid of the `to_ccnf` function Feb 2, 2015 (34fda11e) when I converted the Python implementation of expressions to C.

Can't remember exactly what I was thinking at the time. Probably that a canonical CNF/DNF is necessarily exponential in size, so not that interesting to begin with.

Btw, the code in `boolexpr <http://www.boolexpr.org/index.html>`_ is probably easier to deal with, though I haven't been working on it lately either.

myzinsky, 2023-05-26:
I'm everywhere :-D 

> This comment sent me on a brief trip down memory lane. I got rid of the to_ccnf function Feb 2, 2015 (https://github.com/cjdrake/pyeda/commit/34fda11efcf8f3c61fabcd6af813ae2a3ac46ca9) when I converted the Python implementation of expressions to C.

You should remove it then also from the Docs :-) 

I think it makes sense to still have CCND/CDNF because it is useful for lectures and demonstration purposes.

cjdrake, 2023-05-27:
This is fair.