cjdrake/pyeda #139: Feature Request: Pickle support for ExprNode

SergiyKolesnikov, 2016-11-30:
Currently it is impossible to save a pyeda expression in a file using pickle:
```
test_input_fm = expr('A & B & C')
with open('tmp_test.pickle', 'wb') as tcfm_pickle:
            pickle.dump(test_input_fm, tcfm_pickle)
```
```
      1 test_input_fm = expr('A & B & C')
      2 with open('tmp_test.pickle', 'wb') as tcfm_pickle:
----> 3             pickle.dump(test_input_fm, tcfm_pickle)

PicklingError: Can't pickle <class 'exprnode.ExprNode'>: import of module 'exprnode' failed
```
For long running computations it is often useful to save intermediate results using pickle.

cjdrake, 2016-12-03:
Have you tried the (to_ast([http://pyeda.readthedocs.io/en/latest/reference/boolalg/expr.html#pyeda.boolalg.expr.Expression.to_ast] method, and the (ast2expr)[http://pyeda.readthedocs.io/en/latest/reference/boolalg/expr.html#pyeda.boolalg.expr.ast2expr] functions?

It's not the same as pickling, but it does effectively serialize an expression.

Probably the reason there's a problem here is the `boolexpr` C extension. Pickle must be missing something, and I haven't done any research on it.

SergiyKolesnikov, 2016-12-05:
I am getting a TypeError when I try to convert an AST back to an expression (pyeda 0.28.0):

```
import pyeda.inter as eda
my_expr = eda.expr('A => B')
print(my_expr)
my_expr_ast = my_expr.to_ast()
print(my_expr_ast)
eda.ast2expr(my_expr_ast)
```

**Output:**
```
Implies(A, B)
('impl', ('lit', 1), ('lit', 2))

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-4e111d968d2f> in <module>()
      4 my_expr_ast = my_expr.to_ast()
      5 print(my_expr_ast)
----> 6 eda.ast2expr(my_expr_ast)

/home/kolesnik/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in ast2expr(ast)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/home/kolesnik/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in <listcomp>(.0)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/home/kolesnik/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in ast2expr(ast)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/home/kolesnik/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in <listcomp>(.0)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/home/kolesnik/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in ast2expr(ast)
    230 def ast2expr(ast):
    231     """Convert an abstract syntax tree to an Expression."""
--> 232     if ast[0] == 'const':
    233         return _CONSTS[ast[1]]
    234     elif ast[0] == 'var':

TypeError: 'int' object is not subscriptable
```