cjdrake/pyeda #140: Bug: ast2expr() throws the TypeError (pyeda 0.28.0)

SergiyKolesnikov, 2016-12-08:
pyeda 0.28.0

ast2expr() throws the TypeError:

**Test:**
```
import pyeda.inter as eda
my_expr = eda.expr('A => B')
print(my_expr)
my_expr_ast = my_expr.to_ast()
print(my_expr_ast)
eda.ast2expr(my_expr_ast)
```

**Output:**
```
Implies(A, B)
('impl', ('lit', 1), ('lit', 2))

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-4e111d968d2f> in <module>()
      4 my_expr_ast = my_expr.to_ast()
      5 print(my_expr_ast)
----> 6 eda.ast2expr(my_expr_ast)

/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in ast2expr(ast)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in <listcomp>(.0)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in ast2expr(ast)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in <listcomp>(.0)
    235         return exprvar(ast[1], ast[2])
    236     else:
--> 237         xs = [ast2expr(x) for x in ast[1:]]
    238         return ASTOPS[ast[0]](*xs, simplify=False)
    239 

/anaconda3/lib/python3.5/site-packages/pyeda/boolalg/expr.py in ast2expr(ast)
    230 def ast2expr(ast):
    231     """Convert an abstract syntax tree to an Expression."""
--> 232     if ast[0] == 'const':
    233         return _CONSTS[ast[1]]
    234     elif ast[0] == 'var':

TypeError: 'int' object is not subscriptable
```

cjdrake, 2016-12-11:
The problem is in `exprnodemodule.c`:

```
static const char *ASTOPS[] = {
    "const",
    "lit",

    "or",
    "and",
    "xor",
    "eq",
    "not",
    "impl",
    "ite",
};
```

The `to_ast` method is calling some C code that is incorrectly declaring both variables and complements to both be literals.

cjdrake, 2016-12-11:
Also this bit:

```python
def ast2expr(ast):
    """Convert an abstract syntax tree to an Expression."""
    if ast[0] == 'const':
        return _CONSTS[ast[1]]
    elif ast[0] == 'var':
        return exprvar(ast[1], ast[2])
    else:
        xs = [ast2expr(x) for x in ast[1:]]
        return ASTOPS[ast[0]](*xs, simplify=False)
```

The comparison to `'var'` isn't right; it looks like it should be comparing to `'lit'`.

cjdrake, 2016-12-11:
Also there's the minor problem that the C variable type doesn't have a name, but the Python variable type does. Hrm, deeper and deeper.