cjdrake/pyeda #147: How To See the Graph?!

emomeni, 2018-05-18:
Hi,
I'm trying to have a graphical model of this:

In [1]: %load_ext gvmagic

In [2]: import pyeda

In [3]: from pyeda.inter import *

In [4]: a, b, c = map(bddvar, 'abc')

In [5]: f = a & b | a & c | b & c

In [6]: %dotobj f

In [7]: f.to_dot()

Out[7]: 'graph BDD { n140390891948072 [label=0,shape=box]; n140390892027576 [label=1,shape=box]; n140390840874880 [label="c",shape=circle]; n140390840893680 [label="b",shape=circle]; n140390840906304 [label="b",shape=circle]; n140390840906248 [label="a",shape=circle]; n140390840874880 -- n140390891948072 [label=0,style=dashed]; n140390840874880 -- n140390892027576 [label=1]; n140390840893680 -- n140390891948072 [label=0,style=dashed]; n140390840893680 -- n140390840874880 [label=1]; n140390840906304 -- n140390840874880 [label=0,style=dashed]; n140390840906304 -- n140390892027576 [label=1]; n140390840906248 -- n140390840893680 [label=0,style=dashed]; n140390840906248 -- n140390840906304 [label=1]; }'

this is the example that is available in pyeda documentation, but i didn't see any way to importing this code to jpg/png files!

anyone can help me?
Apparently my graph is now made, but how do I see it? There are no instructions for that on the website

emomeni, 2018-09-20:
Any Idea?!

Abd-Bin, 2023-06-23:
same issue for me