cjdrake/pyeda #152: Building on Windows

alexwebr, 2019-05-03:
Hello,
It seems that PyEDA doesn't easily build on Windows with MSVC, because it uses a C99 feature not available in MSVC.
The error is the following:
```
extension\boolexpr\boolexpr.c(440): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(440): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(440): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(449): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(449): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(449): error C2133: 'xs': unknown size
[... continues ...]
```

It appears the problem is with the use of variable-length arrays in the `BN_OrN`, `BN_AndN`, and related functions:
```
BX_OrN(size_t n, ...)
{
    struct BoolExpr *xs[n];
    READ_ARGS(n, xs);
    return BX_Or(n, xs);
}
```

alexwebr, 2019-05-06:
I opened the PR here: https://github.com/cjdrake/pyeda/pull/153

earphonebreaker, 2022-04-27:
On our server, the gcc doesn't recognize alloca(), so I tried to use malloc instead:
```
struct BoolExpr *
BX_OrN(size_t n, ...)
{
    struct BoolExpr **xs = malloc(sizeof(struct BoolExpr) * n);
    READ_ARGS(n, xs);
    struct BoolExpr* y;
    y = BX_Or(n, xs);
    free(*xs);
    return y;
}
```
It works anyway. Maybe helpful to someone who has the same problem.

closed 2022-07-26 by cjdrake

alexwebr, 2022-07-30:
@earphonebreaker Hi, do you happen to know what version of Linux and GCC this is? If it's something I can get my hands on, I might do a followup to PR #153 to handle this. Thanks!

PR #153 (2022-07-26): Use alloca instead of variable-length arrays to ease builds on Windows
This fixes #152. When installing on Windows through `pip`, this allows the extension to be compiled without errors.