cjdrake/pyeda #164: Windows Installation issues

BenjaminMichaelis, 2021-11-14:
Had problems installing via pip and windows with python 3.10.0 and pip 21.3.1 with ERROR exit status 1. 

Solution I found was downgrading python to 3.9.6 (and subsequently pip is version 21.1.3) and then used the .whl packages to install.

This required
- having the python environment for VS Installed, and possibly [installing the Visual Studio 2012 Redistributable](http://www.microsoft.com/en-us/download/details.aspx?id=30679)
- Downloading the PyEDA .whl file from [Christophe Gohlke’s pythonlibs page](https://www.lfd.uci.edu/~gohlke/pythonlibs/)
- Navigating to the directory where the .whl file is downloaded and running `pip install pyeda-0.28.0-cp38-cp38-win32.whl`  or in my case I ran `py -m pip install pyeda-0.28.0-cp39-cp39-win_amd64.whl`

This worked for myself and multiple others that I am working with! Hopefully this helps!

worseTyler, 2021-11-19:
[For a slightly better link](https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyeda)

Tom-Evers, 2022-11-18:
I don't want to downgrade my python version (I'm on 3.11) but I can't seem to install Pyeda

`pip install pyeda` yields the following errors:
```
boolexpr.c
extension\boolexpr\boolexpr.c(440): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(440): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(440): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(449): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(449): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(449): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(458): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(458): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(458): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(467): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(467): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(476): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(476): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(485): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(485): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(485): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(494): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(494): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(494): error C2133: 'xs': unknown size
extension\boolexpr\boolexpr.c(503): error C2057: expected constant expression
extension\boolexpr\boolexpr.c(503): error C2466: cannot allocate an array of constant size 0
extension\boolexpr\boolexpr.c(503): error C2133: 'xs': unknown size
```

And I can't install any of the (latest) wheels:
`ERROR: pyeda-0.28.0-cp39-cp39-win_amd64.whl is not a supported wheel on this platform.`

asishm, 2022-12-06:
@Tom-Evers you'll have to build from the git repo I believe. The last pypi release was way back in 2015. Gohlke's site won't work since there are no wheels there for py3.10