cjdrake/pyeda #174: AttributeError: module 'collections' has no attribute 'Sequence'

makansij, 2023-04-02:
I couldn't find a good work-around for this:

```
from pyeda.inter import *
X = exprvars('x', 8)
```
Outputs:
```
File /usr/local/lib/python3.10/site-packages/pyeda/boolalg/bfarray.py:430, in farray.__init__(self, objs, shape, ftype)
    429 def __init__(self, objs, shape=None, ftype=None):
--> 430     self._items, autoshape, autoftype = _itemize(objs)
    431     if shape is None:
    432         self.shape = autoshape

File /usr/local/lib/python3.10/site-packages/pyeda/boolalg/bfarray.py:1032, in _itemize(objs)
   1030 def _itemize(objs):
   1031     """Recursive helper function for farray."""
-> 1032     if not isinstance(objs, collections.Sequence):
   1033         raise TypeError("expected a sequence of Function")
   1035     isseq = [isinstance(obj, collections.Sequence) for obj in objs]

AttributeError: module 'collections' has no attribute 'Sequence'
```
I'm using python3.10, and I have installed version 0.28.0.

What can I do to get around this?  Thanks.

b0guslaw, 2023-04-03:
This issue has been reported multiple times and is related to your Python version (See the note that Pyeda has been developed with Python 3.3). To circumvent this you can use
```
import collections
collections.Sequence = collections.abc.Sequence
```

cjdrake, 2023-08-27:
This is fixed now. Whether it has been properly released for Python 3.11 is a separate issue.

closed 2023-08-27 by cjdrake