cjdrake/pyeda #186: Installation Failure on Apple Silicon Machines with latest xcode

jimcarreer, 2024-04-05:
I believe the 3rd party espresso code included in this library is no longer compatible with clang / xcode being installed on Macs:

The following error occurs when simply trying to install the package via pip:
```
[~]$ pip install --no-cache-dir --no-clean pyeda==0.29.0
Collecting pyeda==0.29.0
  Downloading pyeda-0.29.0.tar.gz (486 kB)
     |████████████████████████████████| 486 kB 1.8 MB/s
Using legacy 'setup.py install' for pyeda, since package 'wheel' is not installed.
Installing collected packages: pyeda
    Running setup.py install for pyeda ... error

    < ... snipped for brevity ... >

    running build_ext
    building 'pyeda.boolalg.espresso' extension
    creating build/temp.macosx-13.4-arm64-3.10
    creating build/temp.macosx-13.4-arm64-3.10/pyeda
    creating build/temp.macosx-13.4-arm64-3.10/pyeda/boolalg
    creating build/temp.macosx-13.4-arm64-3.10/thirdparty
    creating build/temp.macosx-13.4-arm64-3.10/thirdparty/espresso
    creating build/temp.macosx-13.4-arm64-3.10/thirdparty/espresso/src
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -Ithirdparty/espresso/src -I/Users/username/.pyenv/versions/3.10.0/envs/test-pyeda/include -I/Users/username/.pyenv/versions/3.10.0/include/python3.10 -c pyeda/boolalg/espressomodule.c -o build/temp.macosx-13.4-arm64-3.10/pyeda/boolalg/espressomodule.o
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -Ithirdparty/espresso/src -I/Users/username/.pyenv/versions/3.10.0/envs/test-pyeda/include -I/Users/username/.pyenv/versions/3.10.0/include/python3.10 -c thirdparty/espresso/src/cofactor.c -o build/temp.macosx-13.4-arm64-3.10/thirdparty/espresso/src/cofactor.o
    thirdparty/espresso/src/cofactor.c:351:50: error: incompatible function pointer types passing 'int (set **, set **)' (aka 'int (unsigned int **, unsigned int **)') to parameter of type 'int (* _Nonnull)(const void *, const void *)' [-Wincompatible-function-pointer-types]
        qsort((char *) (T+2), ncubes, sizeof(set *), d1_order);
                                                     ^~~~~~~~
    /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdlib.h:161:22: note: passing argument to parameter '__compar' here
                int (* _Nonnull __compar)(const void *, const void *));
                                ^
    1 error generated.
    error: command '/usr/bin/clang' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/username/.pyenv/versions/3.10.0/envs/test-pyeda/bin/python3.10 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/44/kk97yyk52hbgkxvry0z928f0hk_94z/T/pip-install-4m2lbrqv/pyeda_fcfdc9e2328549b2b0470997d06f3521/setup.py'"'"'; __file__='"'"'/private/var/folders/44/kk97yyk52hbgkxvry0z928f0hk_94z/T/pip-install-4m2lbrqv/pyeda_fcfdc9e2328549b2b0470997d06f3521/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/44/kk97yyk52hbgkxvry0z928f0hk_94z/T/pip-record-u_ts6obx/install-record.txt --single-version-externally-managed --compile --install-headers /Users/username/.pyenv/versions/3.10.0/envs/test-pyeda/include/site/python3.10/pyeda Check the logs for full command output.
```

Details of the machine this is running on:
```

[~]$ sw_vers
ProductName:        macOS
ProductVersion:     14.4.1
BuildVersion:       23E224
[~]$ clang --version
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
[~]$ xcode-select --version
xcode-select version 2406.
```

Note that I do not know if the machine being Apple Silicon is a contributing factor, this may be broken on x86_64 architectures as well, I just don't have a Mac with that kind of chip to test with.

cjdrake, 2024-04-06:
I don't have that type of Mac either :(.

jist99, 2024-04-08:
I think there has been a warning here for a while and it was promoted to an error later on.  
  
I've found success in working around this by installing `llvm@15` with homebrew and then following the instructions to add it to the .zshrc. Then `clang --version` should show that it's no longer using Apple clang. Using anything later than `llvm@15` brings the issue back. This does mean that it's probably not a MacOS/Apple silicone specific issue but a compiler issue and should be reproducable on other machines.  
  
Since espresso seems to be unmaintained, I think that the fix here involves either fixing the C code yourself or using `espresso-logic` (https://github.com/classabbyamp/espresso-logic), which brought the code for espresso up to a more modern standard. I checked and espresso-logic does not have this issue - though it still generates warnings.

jimcarreer, 2024-04-09:
> I think there has been a warning here for a while and it was promoted to an error later on.
>
> I've found success in working around this by installing `llvm@15` with homebrew and then following the [...]

Yeah I suspected after looking at when the C code was last updated it might have been more a compiler compat issue vs arch.  We weren't really using the functionality supported by pyeda very much to begin with so we were looking at deprecating its use in our code base anyway.

Thanks for the tip though, currently we've only run into this issue locally, but if it is a compiler issue it will eventually hit our docker containers.

Cerebus, 2024-06-14:
Same build error with macOS/x86, FWIW.

Fabs37, 2024-06-15:
I encountered the same problem (Fedora 40, GCC version 14.1.1 20240607).

I fixed it for me by passing `-Wno-incompatible-pointer-types` to the compiler in [setup.py](../blob/main/setup.py#L189), thus making GCC ignoring this specific warning/error.

```python

EXT_MODULES = [
    Extension("pyeda.boolalg.espresso", **ESPRESSO, extra_compile_args=["-Wno-incompatible-pointer-types"]),
    Extension("pyeda.boolalg.exprnode", **EXPRNODE),
    Extension("pyeda.boolalg.picosat", **PICOSAT),
]
```

(I didn't open a PR because it seems kinda quick and dirty and because I'm not at all into developing Python modules.)

ejgroene, 2024-06-19:
Same issue here, on M1.

Since espresso is no longer maintained, I would suggest splitting it off, or at least make it an option. As long as someone maintains pyEDA of course.

I would fix the issue, but then where does the fix go?