cjdrake/pyeda #188: Segmentation fault with satisfy_one function call on a large boolean expression

azewiusz, 2024-06-10:
I do not have much to show as expression is in size of several GB but found out this GDB / Python trace that I would like to understand:
It fails teh same on Windows 10 machine and on Linux one (Ubuntu) _Linux dev 6.8.0-35-generic #35-Ubuntu SMP PREEMPT_DYNAMIC Mon May 20 15:51:52 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux_
```gdb
Thread 1 "python3.11" received signal SIGSEGV, Segmentation fault.
Downloading source file /build/python3.11-khintT/python3.11-3.11.9/build-static/../Objects/obmalloc.c
0x000000000051cd33 in pymalloc_alloc (ctx=<optimized out>, nbytes=32)           
    at ../Objects/obmalloc.c:1979
warning: 1979	../Objects/obmalloc.c: No such file or directory
(gdb) bt
#0  0x000000000051cd33 in pymalloc_alloc (ctx=<optimized out>, nbytes=32)
    at ../Objects/obmalloc.c:1979
#1  _PyObject_Malloc (nbytes=32, ctx=<optimized out>)
    at ../Objects/obmalloc.c:1998
#2  PyMem_Malloc (size=32) at ../Objects/obmalloc.c:623
#3  0x00007ffff7bd8357 in new (ps=ps@entry=0x2cdab00, size=size@entry=32)
    at thirdparty/picosat/picosat.c:949
#4  0x00007ffff7bdb6d6 in new_clause (learned=1, size=1, ps=<optimized out>)
    at thirdparty/picosat/picosat.c:1292
#5  add_simplified_clause (ps=ps@entry=0x2cdab00, learned=learned@entry=1)
    at thirdparty/picosat/picosat.c:2345
#6  0x00007ffff7bda09b in resolve_top_level_unit (reason=0x2cdab70, 
    lit=0x1a1327820, ps=<optimized out>) at thirdparty/picosat/picosat.c:1890
#7  assign_forced (ps=ps@entry=0x2cdab00, lit=0x1a1327820, reason=0x2cdab70)
    at thirdparty/picosat/picosat.c:1975
#8  0x00007ffff7bdbc0d in prop2 (this=0x1a1327825, ps=0x2cdab00)
    at thirdparty/picosat/picosat.c:3895
#9  bcp (ps=ps@entry=0x2cdab00) at thirdparty/picosat/picosat.c:4299
#10 0x00007ffff7be21b7 in bcp (ps=0x2cdab00)
    at thirdparty/picosat/picosat.c:4291
#11 sat (l=-1, ps=0x2cdab00) at thirdparty/picosat/picosat.c:5868
#12 picosat_sat (ps=ps@entry=0x2cdab00, l=-1)
    at thirdparty/picosat/picosat.c:7109
--Type <RET> for more, q to quit, c to continue without paging--
#13 0x00007ffff7bd7dfb in _satisfy_one (self=<optimized out>, 
    args=<optimized out>, kwargs=<optimized out>)
    at pyeda/boolalg/picosatmodule.c:383
#14 0x000000000055394b in cfunction_call (
    func=<built-in method satisfy_one of module object at remote 0x7ffff73938d0>, args=<optimized out>, kwargs=<optimized out>)
    at ../Objects/methodobject.c:542
#15 0x000000000052e6d3 in _PyObject_MakeTpCall (
    tstate=0xa53198 <_PyRuntime+166328>, 
    callable=<built-in method satisfy_one of module object at remote 0x7ffff73938d0>, args=<optimized out>, nargs=<optimized out>, keywords=0x0)
    at ../Objects/call.c:214
#16 0x000000000053bf0d in _PyEval_EvalFrameDefault (tstate=<optimized out>, 
    frame=<optimized out>, throwflag=<optimized out>) at ../Python/ceval.c:4769
#17 0x000000000060eb9d in _PyEval_EvalFrame (throwflag=0, 
    frame=0x7ffff7fb2020, tstate=0xa53198 <_PyRuntime+166328>)
    at ../Include/internal/pycore_ceval.h:73
#18 _PyEval_Vector (tstate=tstate@entry=0xa53198 <_PyRuntime+166328>, 
    func=func@entry=0x7ffff7415ee0, 
    locals=locals@entry={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/dev/Desktop/number_games/sha256/sharun_tse.py') at remote 0x7ffff75fc2d0>, '__spec__': None, '__annotations__': {}, '__builtins__': <module at remote 0x7ffff7588ae0>, '__fi--Type <RET> for more, q to quit, c to continue without paging--
le__': '/home/dev/Desktop/number_games/sha256/sharun_tse.py', '__cached__': None, 'annotations': <_Feature(optional=(3, 7, 0, 'beta', 1), mandatory=None, compiler_flag=16777216) at remote 0x7ffff74fb7d0>, 'dpll': <module at remote 0x7ffff74bcbd0>, 'sha256': <module at remote 0x7ffff5ef56c0>, 'math': <module at remote 0x7ffff74bcf90>, 'pyeda': <module at remote 0x7ffff74bebb0>, 'BooleanTrue': <Singleton(__module__='sympy.logic.boolalg', __doc__='\n    SymPy version of ``True``, a singleton that can be accessed via ``S.true``.\n\n    This is the SymPy version of ``True``, for use in the logic module. The\n    primary advantage of using ``true`` instead of ``True`` is that shorthand Boolean\n    operations like ``~`` and ...(truncated), args=args@entry=0x0, argcount=argcount@entry=0, 
    kwnames=kwnames@entry=0x0) at ../Python/ceval.c:6434
#19 0x000000000060e348 in PyEval_EvalCode (co=<code at remote 0xe16480>, 
    globals=<optimized out>, 
    locals={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/dev/Desktop/number_games/sha256/sharun_tse.py') at remote 0x7ffff75fc2d0>, '__spec__': None, '__annotations__': {}, '__builtins__': <module at remote 0x7ffff7588ae0>, '__file__': '/home/dev/Desktop/number_games/sha256/sharun_tse.py', '__cached__': None, 'annotations': <_Feature(optional=(3, 7, 0, 'beta', 1), mandatory=None, compiler_flag=16777216) at remote 0x7ffff74fb7d0>, 'dpll': <module at remote 0x7ffff74bcbd0>, 'sha256': <module at remote 0x7ffff5ef56c0>, 'math': <module at remote 0x7ffff74bcf90>, 'pyeda': <module at remote 0x7ffff74bebb0>, 'BooleanTrue': <Singleton(__module__='sympy.logic.boolalg', __doc__='\n    SymPy version of ``True``, a singleton--Type <RET> for more, q to quit, c to continue without paging--
 that can be accessed via ``S.true``.\n\n    This is the SymPy version of ``True``, for use in the logic module. The\n    primary advantage of using ``true`` instead of ``True`` is that shorthand Boolean\n    operations like ``~`` and ...(truncated)) at ../Python/ceval.c:1148
#20 0x000000000062d9cc in run_eval_code_obj (
    tstate=0xa53198 <_PyRuntime+166328>, co=0xe16480, 
    globals={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/dev/Desktop/number_games/sha256/sharun_tse.py') at remote 0x7ffff75fc2d0>, '__spec__': None, '__annotations__': {}, '__builtins__': <module at remote 0x7ffff7588ae0>, '__file__': '/home/dev/Desktop/number_games/sha256/sharun_tse.py', '__cached__': None, 'annotations': <_Feature(optional=(3, 7, 0, 'beta', 1), mandatory=None, compiler_flag=16777216) at remote 0x7ffff74fb7d0>, 'dpll': <module at remote 0x7ffff74bcbd0>, 'sha256': <module at remote 0x7ffff5ef56c0>, 'math': <module at remote 0x7ffff74bcf90>, 'pyeda': <module at remote 0x7ffff74bebb0>, 'BooleanTrue': <Singleton(__module__='sympy.logic.boolalg', __doc__='\n    SymPy version of ``True``, a singleton that can be accessed via ``S.true``.\n\n    This is the SymPy version of ``True``, for use in the logic module. The\n    primary advantage of using ``true`` instead of ``True`` is that shorthand Boolean\n    operations like ``~`` and ...(truncated), 
    locals={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/dev/Desktop/number_games/sha256/sharun_tse.py') at remote 0x7ffff75fc2d0>, '__spec__': None, '__annotatio--Type <RET> for more, q to quit, c to continue without paging--
ns__': {}, '__builtins__': <module at remote 0x7ffff7588ae0>, '__file__': '/home/dev/Desktop/number_games/sha256/sharun_tse.py', '__cached__': None, 'annotations': <_Feature(optional=(3, 7, 0, 'beta', 1), mandatory=None, compiler_flag=16777216) at remote 0x7ffff74fb7d0>, 'dpll': <module at remote 0x7ffff74bcbd0>, 'sha256': <module at remote 0x7ffff5ef56c0>, 'math': <module at remote 0x7ffff74bcf90>, 'pyeda': <module at remote 0x7ffff74bebb0>, 'BooleanTrue': <Singleton(__module__='sympy.logic.boolalg', __doc__='\n    SymPy version of ``True``, a singleton that can be accessed via ``S.true``.\n\n    This is the SymPy version of ``True``, for use in the logic module. The\n    primary advantage of using ``true`` instead of ``True`` is that shorthand Boolean\n    operations like ``~`` and ...(truncated)) at ../Python/pythonrun.c:1741
#21 0x0000000000629c66 in run_mod (mod=<optimized out>, 
    filename=<optimized out>, 
    globals={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/dev/Desktop/number_games/sha256/sharun_tse.py') at remote 0x7ffff75fc2d0>, '__spec__': None, '__annotations__': {}, '__builtins__': <module at remote 0x7ffff7588ae0>, '__file__': '/home/dev/Desktop/number_games/sha256/sharun_tse.py', '__cached__': None, 'annotations': <_Feature(optional=(3, 7, 0, 'beta', 1), mandatory=None, compiler_flag=16777216) at remote 0x7ffff74fb7d0>, 'dpll': <module at remote 0x7 [... cut ...]
  <built-in method satisfy_one of module object at remote 0x7ffff73938d0>
  File "/home/dev/Desktop/number_games/venv/lib/python3.11/site-packages/pyeda/boolalg/expr.py", line 1385, in satisfy_one
    return picosat.satisfy_one(self.nvars, self.clauses, assumptions,
  File "/home/dev/Desktop/number_games/venv/lib/python3.11/site-packages/pyeda/boolalg/expr.py", line 738, in satisfy_one
    soln = cnf.satisfy_one(assumptions)
  File "/home/dev/Desktop/number_games/sha256/sharun_tse.py", line ?, in <module>
    (failed to get frame line number)

```

azewiusz, 2024-06-10:
Just to mention it fails for sure on
pyeda 0.29.0

spdOfek, 2026-08-28:
This is solved in the spd-pyeda fork: https://github.com/Speedata-io/pyeda
Uninstall pyeda and `pip install spd-pyeda` should do it.