cjdrake/pyeda #195: Segfault on python version 3.13

ricardo-passthrough, 2025-03-07:
Getting a segfault on the picosat bindings for newer python versions. This works correctly with python3.9 (although I see some instances with complex formulas failing), and fails a lot more often after a python update, even in simple cases. This is an output from my local machine, a macbook with M2 chip, but it also consistently errors on ubuntu and other linux distros.

```python
Python 3.13.2 (main, Feb  4 2025, 14:51:09) [Clang 16.0.0 (clang-1600.0.26.6)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyeda.inter import *
>>> a, b, c, d = map(exprvar, "abcd")
>>> f0 = ~a & b | c & ~d
>>> f0.satisfy_one()
{d: 0, c: 1, b: 0, a: 0}
>>> (a|b).satisfy_one()
[1]    70129 segmentation fault  python3.13 -m pipenv run python
```

I'm trying to debug the c code but I'm a little out of my depth here. Thanks for reviewing!

cjdrake, 2025-03-08:
I appreciate the bug report. I don't do maintenance on this project. Feel free to create a fork with the required fixes.

spdOfek, 2026-08-28:
This is solved in the spd-pyeda fork: https://github.com/Speedata-io/pyeda
Uninstall pyeda and `pip install spd-pyeda` should do it.