cjdrake/pyeda #196: PyEDA fails to build.

matj1, 2025-04-08:
If I try to install PyEDA with `pip install pyeda`, it fails with this error:

```
thirdparty/espresso/src/cofactor.c:351:50: error: passing argument 4 of ‘qsort’ from incompatible pointer type [-Wincompatible-pointer-types]
  351 |     qsort((char *) (T+2), ncubes, sizeof(set *), d1_order);
      |                                                  ^~~~~~~~
      |                                                  |
      |                                                  int (*)(set **, set **) {aka int (*)(unsigned int **, unsigned int **)}
In file included from thirdparty/espresso/src/utility.h:6,
                 from thirdparty/espresso/src/espresso.h:8,
                 from thirdparty/espresso/src/cofactor.c:3:
/usr/include/stdlib.h:971:34: note: expected ‘__compar_fn_t’ {aka ‘int (*)(const void *, const void *)’} but argument is of type ‘int (*)(set **, set **)’ {aka ‘int (*)(unsigned int **, unsigned int **)’}
  971 |                    __compar_fn_t __compar) __nonnull ((1, 4));
      |                    ~~~~~~~~~~~~~~^~~~~~~~
error: command '/usr/bin/gcc' failed with exit code 1
```

I use Python 3.13.2 on Arch Linux.

<details><summary>The full report</summary>

```
Collecting pyeda
  Using cached pyeda-0.29.0.tar.gz (486 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pyeda
  Building wheel for pyeda (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pyeda (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [82 lines of output]
      /tmp/pip-build-env-59e_gu21/overlay/lib/python3.13/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!
      
              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:
      
              License :: OSI Approved :: BSD License
      
              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************
      
      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-313/pyeda
      copying pyeda/util.py -> build/lib.linux-x86_64-cpython-313/pyeda
      copying pyeda/inter.py -> build/lib.linux-x86_64-cpython-313/pyeda
      copying pyeda/__init__.py -> build/lib.linux-x86_64-cpython-313/pyeda
      creating build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/table.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/minimization.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/expr.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/boolfunc.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/bfarray.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/bdd.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      copying pyeda/boolalg/__init__.py -> build/lib.linux-x86_64-cpython-313/pyeda/boolalg
      creating build/lib.linux-x86_64-cpython-313/pyeda/logic
      copying pyeda/logic/sudoku.py -> build/lib.linux-x86_64-cpython-313/pyeda/logic
      copying pyeda/logic/graycode.py -> build/lib.linux-x86_64-cpython-313/pyeda/logic
      copying pyeda/logic/aes.py -> build/lib.linux-x86_64-cpython-313/pyeda/logic
      copying pyeda/logic/addition.py -> build/lib.linux-x86_64-cpython-313/pyeda/logic
      copying pyeda/logic/__init__.py -> build/lib.linux-x86_64-cpython-313/pyeda/logic
      creating build/lib.linux-x86_64-cpython-313/pyeda/parsing
      copying pyeda/parsing/token.py -> build/lib.linux-x86_64-cpython-313/pyeda/parsing
[... 30 lines cut ...]
      creating build/temp.linux-x86_64-cpython-313/thirdparty/espresso/src
      gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -ffat-lto-objects -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -fPIC -Ithirdparty/espresso/src -I/home/matj1/trappist-fork-venv/include -I/usr/include/python3.13 -c pyeda/boolalg/espressomodule.c -o build/temp.linux-x86_64-cpython-313/pyeda/boolalg/espressomodule.o
      gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -ffat-lto-objects -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -fPIC -Ithirdparty/espresso/src -I/home/matj1/trappist-fork-venv/include -I/usr/include/python3.13 -c thirdparty/espresso/src/cofactor.c -o build/temp.linux-x86_64-cpython-313/thirdparty/espresso/src/cofactor.o
      thirdparty/espresso/src/cofactor.c: In function ‘simplify_cubelist’:
      thirdparty/espresso/src/cofactor.c:351:50: error: passing argument 4 of ‘qsort’ from incompatible pointer type [-Wincompatible-pointer-types]
        351 |     qsort((char *) (T+2), ncubes, sizeof(set *), d1_order);
            |                                                  ^~~~~~~~
            |                                                  |
            |                                                  int (*)(set **, set **) {aka int (*)(unsigned int **, unsigned int **)}
      In file included from thirdparty/espresso/src/utility.h:6,
                       from thirdparty/espresso/src/espresso.h:8,
                       from thirdparty/espresso/src/cofactor.c:3:
      /usr/include/stdlib.h:971:34: note: expected ‘__compar_fn_t’ {aka ‘int (*)(const void *, const void *)’} but argument is of type ‘int (*)(set **, set **)’ {aka ‘int (*)(unsigned int **, unsigned int **)’}
        971 |                    __compar_fn_t __compar) __nonnull ((1, 4));
            |                    ~~~~~~~~~~~~~~^~~~~~~~
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyeda
Failed to build pyeda
ERROR: Failed to build installable wheels for some pyproject.toml based projects (pyeda)
```

</details>

matj1, 2025-04-08:
The compilation works with Python 3.10 (on a different system). So I think that the failure is caused by the Python version.

spdOfek, 2026-08-28:
This is solved in the spd-pyeda fork: https://github.com/Speedata-io/pyeda
Uninstall pyeda and `pip install spd-pyeda` should do it.